microsoft/AirSim #1049: Modify position of the camera on drone

showaykerker, 2018-05-03:
I realize that there's an API to change orientations of the already-launch cameras on drone, and also know that  it's easy to add additional cameras via setting.json, however, after navigating source codes, I found it hard to understand how to modify cameras' position.
Is there a way to, for example, move the camera with camera_id = 1 to the right of the drone ?

btw, I'm working on a project which needs to get 360 degree point cloud around the vehicle. Are there any chances that a lidar plugin will be release in the future? Or is there any recommendation on how I can implement it ?

sytelus, 2018-05-03:
You can change existing camera positions here: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L43

closed 2018-05-08 by sytelus

SAmmarAbbas, 2018-06-23:
@sytelus I want to modify the camera position and orientation but the above link is not working. Could you check it again? Thanks

showaykerker, 2018-06-23:
@SAmmarAbbas 
I believe it's here 
https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Vehicles/Car/CarPawn.cpp#L43

Beijingsheng, 2018-08-03:
I have changed the numbers in the FVector of front center camera, however when I use this camera to get images, it seems like the camera is still at its default position. What might cause this？

Omar-Elmofty, 2020-06-27:
I have the same question, I am wondering what did you do to be able to modify the camera's position live while the drone is flying?