microsoft/AirSim #1052: RC at index 0 and index 1 issue + Axis mapping problem

haider04, 2018-05-04:
Hi,

I am using steering wheel (Leobodnar) and Club support Pedals (Fanatec) from 2 different manufacturers. When i plug in the USB (1 for wheel and 1 for pedals) and open Game controller options, i can see my wheel showing at top (index 0) and pedals at 2nd place (index 1). I am using below settings in settings.json file:
{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.0,
"SimMode" : "Car",
"PhysXCar" : {
"RC": {
"RemoteControlID": 0
}
}
} 
But the problem is, when i connect both steering wheel and pedals, only steering wheel change axis values because of index 0, and pedals don't have any effect. And when i unplug the steering wheel and connect only the pedals, pedal axis (throttle and brake) change the values because now pedals take index 0. I saw the change in axis values by debugging ACarPawn.cpp file.

Now i am facing 2 problems:

1. How to connect both steering wheel and axis so they are both detected in AirSim? Can i use another "RemoteControlID": 1  in settings.json file along with ID 0? If not, what could be the possible solution? 

2. My wheel and throttle both are mapped on the same axis, left_x axis (i observed the value change in left_x axis for both wheel and throttle, while debugging ACarPawn.cpp). Is there any possibility to change the axis mapping of wheel and pedals? Because when i press throttle, the car steer and accelerates at the same time. 

Can someone please help me out to solve these problems? Suggestions, guidance, experiences are highly appreciated. Thanks!

Regards,
Haider

sytelus, 2018-05-08:
Current code can only deal with one RemoteControlID. For Logitech G920, wheels and pads comes through same USB port so that works out fine. However it might be good idea to support pads on different USB port. For this to work, you will probably need to add new setting, for example, "AuxRemoteControlID". Then in SimJoyStick.cpp, you can have two instances of DirectInputJoyStick talking to two different ports. Then you can use VendorID to detect which one is which and setup final outputs.

That's rough outline of code changes you will need. Please feel free to contribute!

haider04, 2018-05-17:
Thanks for your suggestion. What if i use vJoy and UCR (Universal control Remapper) to make my both indices work simultaneously as one? Will it work?

haider04, 2018-05-30:
Hi Sytelus,
Where should i add the new settings to add another remote control ID? In Settings.hpp or AirSimSettings.hpp? And how parse it into the code?