microsoft/AirSim #1053: No change occurring in PID controller

ReshadZadran, 2018-05-04:
Hey guys,
I have tried to change the values kp, ki and kd for the pid controller. Nothing noticeable happened in the simulation. I had hoped the drone would oscillate or behave different than normal. I tried to change the pid values in: 
Documents\AirSim\AirLib\include\controllers\PidController.hpp,
Documents\AirSim\AirLib\include\vehicles\multirotor\firmwares\simple_flight\firmware\interfaces\ CommonStructs.hpp,
Documents\AirSim\AirLib\include\vehicles\multirotor\firmwares\simple_flight\firmware\ PidController.hpp,
Documents\AirSim\AirLib\include\vehicles\multirotor\firmwares\simple_flight\firmware\ StdPidIntegrator.hpp.

What I want to do, is to change the values for ki, kp and kd to some desired value that I choose, for the drone given by AirSim. I expect a change in the simulation when the pid values have been changed. Is this possible? If so, were can I find the code which I need to edit? Help is much appreciated, thanks.

sytelus, 2018-05-08:
Are you running Blocks project in VS2017? Then you should be aware of this gotcha: https://github.com/Microsoft/AirSim/blob/master/docs/dev_workflow.md#i-made-changes-in-code-in-blocks-project-but-its-not-working

sytelus, 2018-05-14:
Yes, just add line and the delete it. That makes file dirty and tells build system to do full recompile. Alternatively you can click Rebuild in Build menu in VS 2017.

closed 2018-05-15 by ReshadZadran

sandilyasg, 2024-02-06:
@sytelus @Resh404 how would I do this with Unity? Do I change the hpp files and then rebuild?