microsoft/AirSim #1062: An interesting thing NH environment.

wujiyoung, 2018-05-05:
I downloaded the latest release assets AirSimNH and CityEnviron, I tried to capture `scene`, `depthPerspective`, `depthVis`, and `segmentation` in both environment. I used the same settings:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "SimMode": "Car",
  "Recording": {
    "RecordOnMove": false,
    "Cameras": [
        { "CameraID": 0, 
          "ImageType": 0, 
          "PixelsAsFloat": false, 
          "Compress": true 
        },
        { "CameraID": 0, 
          "ImageType": 2, 
          "PixelsAsFloat": true, 
          "Compress": true 
        },
        { "CameraID": 0, 
          "ImageType": 3, 
          "PixelsAsFloat": false, 
          "Compress": true 
        },
        { "CameraID": 0, 
          "ImageType": 5, 
          "PixelsAsFloat": false, 
          "Compress": true 
        }
    ]
  },
  "CaptureSettings": [
    {
      "ImageType": 0,
      "FOV_Degrees": 120,
      "Width": 640,
      "Height": 480
    },
    {
      "ImageType": 2,
      "FOV_Degrees": 120,
      "Width": 640,
      "Height": 480
    },
    {
      "ImageType": 3,
      "FOV_Degrees": 120,
      "Width": 640,
      "Height": 480
    },
    {
      "ImageType": 5,
      "FOV_Degrees": 120,
      "Width": 640,
      "Height": 480
    }
  ]
}
```
Everything was Ok in City environment, while in NH environment, I found that the `scene` did not match the `segmentation`, the `depthVis` was also not correct. Can someone give me a explanation, or is it a bug ?
[image]
[image]
[image]

showaykerker, 2018-05-07:
Same problem here.
But in my opinion, it's not actually binary image, do np.unique then you will get something like this [  0   1   2   3   4   5   6 251 252 253 254 255]. 
It's not the correct value though, I think it's because of the transition between byte data returns by simGetImage and imdecode, but i've got no idea how to decode it into float32 or float64. 
I've already set PixelsAsFloat to true.

sytelus, 2018-05-08:
See here for how to get floats: https://github.com/Microsoft/AirSim/blob/master/PythonClient/hello_car.py#L57

Saving and visualizing depth image as float is hard because RGB only allows up to 255.

closed 2018-05-08 by sytelus

wujiyoung, 2018-05-09:
@sytelus Thank you for your response.
Can you give further explaination that why the `segmentation` image didn't match the `scene` image?