microsoft/AirSim #1063: Access keyboard input with C++ api

wujiyoung, 2018-05-05:
I want to start recording when a keyboard is pressed in my C++ project, just as the official releases do. But I don't know how to archive the keyboad input in my code, is there any advice?

Robin-Wong, 2018-05-07:
I am using windows, so for keyboard:
        while (1)
        {
            if (GetAsyncKeyState('G') & 0x1)
            {
                std::cout << "Press: G..." << std::endl;
                startGetImage = 1;
                pause = 0;
            }
            if (GetAsyncKeyState('P') & 0x1)
            {
                std::cout << "Press: P..." << std::endl;
                pause = 1;
            }
        }

wujiyoung, 2018-05-07:
@Robin-Wong Thank you, it workd for accessing keyboard input. Following is my demo code:
```
while(true)
{
  if ((GetAsyncKeyState('S') & 0x1) || (GetAsyncKeyState('s') & 0x1))
  {
    isCapture = !isCapture;
    std::cout << "S/s is pressed, recording state: " << isCapture << std::endl;
    if (isCapture)
    {
      client.simPrintLogMessage("start recording");
    }
    else
    {
      client.simPrintLogMessage("stop recording");
    }
  }
  if (isCapture)
  {
    saveImage(client, "images/");
  }
}
```
However, their was another problem. I could only read the input key '`s/S`' for the first time, after then, isCapture was always true. It seemed that the function `GetAsyncKeyState('S')` was not asynchronous, function `saveImage(client, "images/")` was only excuted at last.

wujiyoung, 2018-05-07:
I used the `client.simGetImages(request)` in function `saveImage(client, "images/")`, does this matter?

wujiyoung, 2018-05-07:
I found the reason. I was so foolish, I added a standard input `cin` in function `saveImage(client, "images/")`

sytelus, 2018-05-08:
For Unreal related question, please use http://answers.unrealengine.com

closed 2018-05-08 by sytelus