microsoft/AirSim #1064: Radial pattern in DepthPlanner Images

Mayankm96, 2018-05-05:
Hey all! 

I have been collecting the `DepthPlanner` images at 720p resolution. Although the rendering of the images takes time, the issue I am facing is that a radial pattern is observed when I convert the pfm images into the heat map as shown here. I would like to know if this is an error due to the heatmap conversion or the `DepthPlanner` images themselves are incorrect? 

[image]

sytelus, 2018-05-08:
It's hard to tell without debugging through your code... You might want to look at float values in depth images and see if that is the problem. Discretization somewhere in code can also be issue.

Mayankm96, 2018-05-10:
I updated the AirSim plugin to the most recent one with VS2017 and Unreal 4.18. The issue seems to have been fixed. I'll reopen the issue if it resurfaces again. Thank you! :)

[image]

Mayankm96, 2018-06-18:
I collected the depthVis images this time and observed that a radial pattern exists in them now. 

Heatmap from the depthVis images:
[image]

Heatmap from depthPlannar images:
[image]

Any idea with what's the issue?

sytelus, 2018-10-17:
DepthViz used to do cycles at every 100m distance and these rings very purposeful for visualization.

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus