microsoft/AirSim #1065: Fixing camera position and angle in Game Mode

kshitijgoel007, 2018-05-06:
HI!

Thanks for keeping various modes for the following camera in game mode. I am trying to fix the camera in position and orientation, instead of moving it with the vehicle. Is it possible in the current API? I think the manual mode (activated by pressing 'M' in game mode) is the closest to this point, but that too follows the vehicle. 

Thanks in advance!

Kshitij

Robin-Wong, 2018-05-07:
The modify on camera's orientation must in Game Thread.
I using the code
UAirBlueprintLib::RunCommandOnGameThread([&param, camera]() { cameta->SetActorRotation(param); }, true);
for modify the camera's orientation in my own thread.

husha1993, 2018-05-08:
https://github.com/Microsoft/AirSim/blob/d088719d0b4bf19901e96045581c21177158816d/PythonClient/AirSimClient.py#L500

We can set the camera orientation relative to the drone body frame.
But I haven't found the api for set the camera position yet.

sytelus, 2018-05-08:
Manual view mode shouldn't move the camera. Are you using Computer Vision mode, drone or car?

kshitijgoel007, 2018-05-18:
@sytelus I am trying to create support for this (and corresponding pull request). Can you provide me some background on how the manual view camera works?