microsoft/AirSim #1066: setting.json can not be changed(be rewritten everytime)

allen9273, 2018-05-07:
The path of the setting.json is C:\Users\chen9\OneDrive\文档\AirSim.
win10 Home
vs2015 pro3
The problem is only produced in my laptop. My desktop is fine. No matter what I change in setting.json, whenever I run the binary (for example AirsimNH), it would be written as default value like this:
[image]
Could anyone solve this problem?

sytelus, 2018-05-08:
I think this is probably happening due to asian characters in file path. Unfortunately we can't reproduce this issue on English versions. You can help s debug this, however. The relevant line of code is here:

https://github.com/Microsoft/AirSim/blob/7bddd5857791f9c164e8eba80c229f199c0babf8/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L270

allen9273, 2018-05-09:
Thanks for your reply.  I have found the solution. Because of the Asian characters in file path, just right click on the "Documents"-"Properties" - "Location" , you will find the path content Asian characters. Just modify it to English, click on "OK". Everything will be fine.
[image]
[image]

sytelus, 2018-05-14:
There might be side effect to this because you are renaming you document folder. For example, other applications which has set its path to location before you renamed might not work properly.

Beijingsheng, 2018-08-08:
This is the path of my settings.json C:\Users\risheng\Documents\AirSim. Still the problem exists.

Beijingsheng, 2018-08-08:
The Users folder is currently showing 用户, however it is Users when I check the path. I tried to change 用户 but it's a default folder and its name can't be changed.

allen9273, 2018-08-08:
[image]
Can you copy your path  C:\Users\risheng\Documents\AirSim here and enter, check if the folder content settings?

Beijingsheng, 2018-08-08:
It works, in fact I just changed the display language from Chinese to English so I suppose there shouldn't be any Chinese character in the path?

madratman, 2019-01-23:
Yeah, chinese characters have caused errors reported in a few more issues as well. Let's stick to English. 
Closing this as it's resolved. please comment/re-open as needed.

closed 2019-01-23 by madratman

damilolah, 2019-08-16:
I use English and still have the same issue my file part is: C:\Users\4567\Documents\AirSim

Please help!!!

allen9273, 2019-08-17:
[image]
copy your path C:\Users\4567\Documents\AirSim here and enter, check if the folder content settings?

andreasceid, 2021-05-16:
Greek characters don't work too. That gives me an impression that all non Latin characters trigger this reset to the `settings.json` file. Path of `settings.json` file to simulate the problem was: `C:\Users\example_user\Έγγραφα`. The solution provided above applies for Greek characters, too.