microsoft/AirSim #1069: Block Environment build error

EthanZhangYi, 2018-05-07:
I pull the latest master branch and build it. Then I follow [this instruction](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md) to run the build-in block environment. And error occurs.
[image]
Window 10
Visual studio 2017 Community Version
Unreal 4.18.3

Also, it seems that the block environment is not updated for VS2017
[image]

sytelus, 2018-05-08:
There was one minor error in docs which I just fixed: Please use VS 2017 (not VS2015). Then select Blocks as Startup Project by right clicking in VS.

vinbo, 2018-06-12:
so even the name of "Block" is followed by(visual studio 2015), it should affect nothing, right?

msb336, 2019-10-03:
This should be resolved in the current master. please re-open if you are still having issues.

closed 2019-10-03 by msb336