microsoft/AirSim #1071: Camera pixel size in Unreal Units

nikolaid77, 2018-05-07:
What is the camera pixel size in Unreal Units?

sytelus, 2018-05-08:
Can't understand your question... camera pixel in unreal has no "size" and its not related to UU.

closed 2018-05-08 by sytelus

nikolaid77, 2018-05-09:
@sytelus : I am novice in airsim/unreal and I respect your opinion. However, after posting the question  I have searched a bit on the topic. In the [cinecameraactor](https://docs.unrealengine.com/en-us/Engine/Sequencer/HowTo/CineCameraActors) more specifically in the filmback settings, one can select manually the sensor size (width / height) in mm. This, in combination with the sensor resolution  (width / height in pixels) implies that there is a pixel size, in mm. Let me know if I am wrong on the above. If the above are correct regarding the cine camera, then the question can be rephrased as: how can I know or calculate the pixel size in the "simple" camera.

Zartris, 2021-01-26:
Did we ever get an answer to this?

vyunolbek, 2024-03-13:
> @sytelus : I am novice in airsim/unreal and I respect your opinion. However, after posting the question I have searched a bit on the topic. In the [...]

Have you received an answer to your question?