microsoft/AirSim #1074: Why the 'kinematics_true' and 'kinematics_estimated' are exactly the same?

husha1993, 2018-05-08:
Hi,

I tried to set the drone in different position and orientation, and I found in all the cases the 'kinematics_true' and 'kinematics_estimated'  are exactly the same which seems unreasonable. 

According to my understanding to the paper, the kinematics_true is obtained from the physics engine and the kinematics_estimated is estimated from the flight_control, they should be different.

Domi47, 2018-05-08:
Hi, as it's said in https://github.com/Microsoft/AirSim/blob/master/docs/apis.md : "Please note that simple_slight currently doesn't support state estimator which means estimated and ground truth kinematics values would be same for simple_flight. Estimated kinematics are however available for PX4 except for angular acceleration."

husha1993, 2018-05-08:
I see, thanks!

closed 2018-05-08 by husha1993