microsoft/AirSim #1075: Package Error in Linux

JenaEmz, 2018-05-08:
os:ubuntu16.04
UE4:4.18
I meet an error when i package blocks.uproject,how can i solve it?guys.
[9/23] Compile Module.InputDevice.cpp
  In file included from /home/jena/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/Blocks/Development/AirSim/Module.AirSim.gen.cpp:2:
  In file included from /home/jena/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/Blocks/Inc/AirSim/AirBlueprintLib.gen.cpp:8:
  ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:140:58: error: member access into incomplete type 'UWorld'
          APlayerController* controller = actor->GetWorld()->GetFirstPlayerController();
                                                           ^
  Runtime/Core/Public/GenericPlatform/GenericPlatformMisc.h:817:25: note: forward declaration of 'UWorld'
          static bool Exec(class UWorld* InWorld, const TCHAR* Cmd, FOutputDevice& Out)
                                 ^
  In file included from /home/jena/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/Blocks/Development/AirSim/Module.AirSim.cpp:2:
  In file included from /home/jena/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.cpp:4:
  ../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirBlueprintLib.h:140:58: error: member access into incomplete type 'UWorld'
          APlayerController* controller = actor->GetWorld()->GetFirstPlayerController();
                                                           ^
  Runtime/Core/Public/GenericPlatform/GenericPlatformMisc.h:817:25: note: forward declaration of 'UWorld'
          static bool Exec(class UWorld* InWorld, const TCHAR* Cmd, FOutputDevice& Out)
                                 ^
  [10/23] Compile Module.DatabaseSupport.cpp
  [11/23] Compile Module.Advertising.cpp

kshitiz38, 2018-05-31:
just add this to your header file (AirBlueprintLib.h)
`#include "Runtime/Engine/Classes/Engine/World.h"`
[source](https://answers.unrealengine.com/questions/672972/build-for-html-error.html)

AloshkaD, 2018-06-07:
@kshitiz38  that worked for me, thank you!

TritonSailor, 2018-06-11:
@kshitiz38 That solved the issue for me.

closed 2019-10-03 by msb336