microsoft/AirSim #1076: confusion on gathering training data for deeplearning

wujiyoung, 2018-05-08:
I am struggling to access training data for depth prediction in car mode. Now I can only drive the car manually to capture the `scene` and `depthVis`. I believe this is a disturbing and time-consuming work. Therefore, I am trying to have my car driving along a path automatically, but it is difficult since we can not give a proper path, we even don't have a global map.
Any one esle has ever used AirSim to gather training data? Do you gather data by manually driving too?

sytelus, 2018-05-08:
You can use [Computer Vision mode](https://github.com/Microsoft/AirSim/blob/v1.1.10/docs/image_apis.md#computer-vision-mode) and use setSimPose API to set camera on points on path you want. Alternatively, you can write PID controller to keep car on certain path but that's probably unnecessary for your purpose.

closed 2018-05-08 by sytelus

wujiyoung, 2018-05-09:
Thank you @sytelus, I do have used Computer Vision mode.
However, there is no function `setSimPose` in `CarClient` in python API, neither functions like `moveOnPath`, `moveToPosition`.