microsoft/AirSim #1077: Is it possible to modified vehicle dynamics with another UnrealPlugin --Carsim?

DrifterFun, 2018-05-08:
Hi
I am wondering that is it possible to change the Vehicle dynamics equations and use a model from Carsim? (Another plugin in Unreal?)
https://www.unrealengine.com/marketplace/carsim-vehicle-dynamics
It seems that Carsim may have more accurate model because it has worked on car simulation for many years.
Thanks!

sytelus, 2018-05-08:
You certainly can. However you will need to change CarPawn class. You can help us contribute this feature :).

closed 2018-05-08 by sytelus

DrifterFun, 2018-06-03:
Dear Community,
I have finish the works talked above. I will later fork AirSim to my own  repository. And I will give usage in my own blog. The Link I will update here few days later.
Thanks!

freeyyc, 2019-09-23:
> Dear Community,
> I have finish the works talked above. I will later fork AirSim to my own repository. And I will give usage in my own blog. The Link I will update here few days later.
> Thanks!

Can you share some detail with us ?