microsoft/AirSim #1088: Disable collision checking in CV Mode

kshitijgoel007, 2018-05-12:
I am trying to disable collision checking, both between multiple pawns and pawn with the environment. I am currently using Python Client and running all the vehicles in CV Mode. How do I disable all the collision checking functionality? (I have my own collision checker that I want to test)

kshitijgoel007, 2018-05-14:
@sytelus Works, thanks. Closing.

patrickschulte, 2018-05-25:
I got the same issue. Try to change positions/poses, but it pawn with the environment. The 

`"EnableCollisionPassthrogh": true` command did not help!

kshitijgoel007, 2018-06-27:
@sytelus Is there any update on this issue?

liqimai, 2023-12-15:
@sytelus Is there any update on this issue? I want to disable collisions between multiple vehicles but keep collisions with the environment, so that I can simulate tons of cars at the same time to speed up RL training.

closed 2025-07-04 by kshitijgoel007