microsoft/AirSim #1096: Getting 3D (+ projected) coordinates within camera coordinate system

phildue, 2018-05-16:
Hi,

I would like to get the coordinates of an object / any coordinate in 3D world space in coordinates with respect to a certain camera. Subsequently I would like to project these coordinates to the image plane of a that camera.

What would be the best way to achieve this? I feel this would be something generally useful so I'm also happy to contribute by filing a pull request.

My approach so far is getting the coordinates of the camera and the object with getCameraInfo() and getObjectPose() and creating the rotation/trainslation matrices Rt_cam and Rt_object. The 3D coordinates within the camera reference frame I get with inv(Rt_cam).dot(Rt_object). 

Subsequently I would estimate the intrinsic camera matrix based on the FOV and the resolution and do a perspective projection.

The first step is working most of the time but sometimes not so I'm wondering if my approach is generally the right way or if I have to do something different. Any help would be appreciated :)

kdzy, 2019-03-18:
@phildue Hi, 
Have you ever solved this issue?  I want to get the object position in world 3D space from image pixel and  I encountered the similar problem. As far as I'm concerned, the camera model in AirSim is not the classic pinhole camera model, but what exactly the camera model is I  still don't clear yet ? Have you found out the issues? I would appreciate it if you could communicate and share with me.

phildue, 2019-03-18:
Hi @kdzy ,
yes I solved this. AirSim is using a pinhole camera. What took me a while to realize is that the there is a difference between the position of the drone and the position of the camera. This you have to take into account when applying the transformations. You can checkout [my repo here](https://github.com/phildue/datagen) where I implemented it.

kdzy, 2019-03-18:
@phildue Thanks for your quick reply.
How do you confirm  AirSim is using a pinhole camera? And what is the camera's intrinsic parameter？AirSim uses UE4's camera model directly. I have checked the relevant literature, but did not find the camera model description of UE4. In addition, I have tried to use pinhole  model to determine the object position, but the error is very large， and I also considered the camera and the drone different positions.  I wonder if I can get some clues from you.

phildue, 2019-03-18:
Well in the end it worked for me using a pinhole model. From that I concluded they use one. Also I'm not aware of other suitable models. (But I also don't have strong background in computer graphics).  The parameters you can compute from image resolution and fov. There are also other issues in this from where they estimate it by placing a chessboard with UE4Editor. Just search through the issues.

kdzy, 2019-03-19:
Ok, Thanks. I have got it. I will have a try

closed 2020-02-14 by ironclownfish