microsoft/AirSim #1099: New type of vehicle possible ? (Boat)

Fractaos, 2018-05-16:
I would like to use AirSim to simulate and train a water based drone, I've a water simulation with buoyancy and I manage to make the car float and change it's model but can't make it move (because of the wheel which aren't in contact with anything). Another idea was to do the same with the flying drone, but this one can't float, he just fall off until he it the floor. 
For a first shot I don't need something very realistic, just a props that can float and move and for now, I can't achieve it, mainly by lack of understanding of how the "drones" in AirSim are built and works. 

Thank's if you go a solution !

TritonSailor, 2018-09-09:
@Fractaos which water simulation are you using?

Fractaos, 2019-02-16:
Hello, @TritonSailor 

Sorry for the time without answering. I'm currently not on the project but I can still answer you, it was this one : [OceanProject](https://github.com/UE4-OceanProject/OceanProject)

Hope it could be usefull in the futur.

closed 2020-02-14 by ironclownfish