microsoft/AirSim #1100: Error in building AirSim: "No such file or directory"

MohamedAliRashad, 2018-05-16:
I get this error when i run ```./build.sh```
 

```
putting build in build_debug folder, to clean, just delete the directory
The C compiler identification is Clang 5.0.0
-- The CXX compiler identification is Clang 5.0.0
-- Check for working C compiler: /usr/bin/clang-5.0
-- Check for working C compiler: /usr/bin/clang-5.0 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-5.0
-- Check for working CXX compiler: /usr/bin/clang++-5.0 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Running CommonSetup...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED
-- Performing Test UNUSED_LAMBDA_CAPTURE_WARN_SUPPORTED - Failed
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED
-- Performing Test ZERO_AS_NULL_POINTER_CONSTANT_WARN_SUPPORTED - Failed
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Running CommonSetup...
-- found AIRSIM_ROOT=/home/mohamed/Work/GitHub Projects/AirSim
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mohamed/Work/GitHub Projects/AirSim/build_debug
Scanning dependencies of target rpc
[  1%] Building CXX object rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o
clang: error: no such file or directory: 'Projects/AirSim/llvm-build/output/include/c++/v1'
rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/build.make:62: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o' failed
make[2]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/lib/rpc/dispatcher.cc.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all' failed
make[1]: *** [rpclib_wrapper/rpclib/CMakeFiles/rpc.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

abhimanyuchadha96, 2018-10-25:
I am running into the same error.

DrDumbenstein, 2019-05-20:
same problem is it solved?

J-Schaefer, 2019-10-18:
I encountered the same issue. The path at which the compiler is looking for starts with "Projects/...". See at the end `clang: error: no such file or directory: 'Projects/AirSim/llvm-build/output/include/c++/v1'`.
Gave me the idea that the path is splitted falsely. So apparently UnrealEngine creates the folder `Unreal Projects` **with** a space in between as default project folder. I removed the space and it worked.

mmtalaie, 2019-10-18:
I have the same problem on Ubuntu 18.
is it solve?