microsoft/AirSim #1106: Supported Unreal version on Linux?

gmosem, 2018-05-21:
Hi,
in the installation instructions for Linux a note says that online Unreal 4.17 is supported.
However in the command line instructions is shown:
`git clone -b 4.18 https://github.com/EpicGames/UnrealEngine.git`
That means we should download the 4.18 version.

The question is, which is the last supported version of Unreal? 4.17 or 4.18 ?
Thanks!

dileep-krishnan, 2018-05-22:
Hi!
They have changed the Unreal version to 4.18. I also got confused at first and tried to use 4.17 in Linux, but didn't work out. So, I used Unreal 4.18 and it works. The error is in the documentation, it needs to be changed.

mantkiew, 2018-06-05:
Also, are there plans for supporting 4.19 anytime soon?

TritonSailor, 2018-06-05:
I've been using 4.19.2 on windows and linux, haven't experienced any issues.

sytelus, 2018-06-20:
Doc is fixed now. Looks like 4.19 should pretty much just work.

closed 2018-06-20 by sytelus