microsoft/AirSim #1110: City Environment project

petroolg, 2018-05-23:
Hi, 
I've successfully built UE and AirSim on Linux following docs and now I'd like to setup my own environment for training an RL agent.  I'd like to have City environment (large environment with moving vehicles and pedestrians), the one from prebuilt envs. However I cannot find it on Unreal marketplace or anywhere. 
Where can I possibly get it?

kshitijgoel007, 2018-05-27:
I use modular neighborhood pack https://www.unrealengine.com/marketplace/modular-neighborhood-pack

kshitijgoel007, 2018-05-27:
You can use this doc https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md to set up for custom environment. However, first check that all of the plugin functionality that you want works fine in the supplied Blocks environment.

Domi47, 2018-05-28:
Hi @kshitijgoel007, I have a question, does Modular Neighborhood Pack allow you to fly inside buildings?

Thanks!

kshitijgoel007, 2018-05-28:
Hi @Domi47, I tried doing that. You would either have do delete or tilt the doors in the buildings that you want to enter. Also, you might want to disable the collision checker as well, refer #1088 for updates on that.

sytelus, 2018-06-20:
We are also planning to create binaries of these environments for Linux soon.

closed 2018-06-20 by sytelus

kshitijgoel007, 2018-09-27:
@mallela I am assuming that you are talking about `AirSimNH.zip` listed here: https://github.com/Microsoft/AirSim/releases. I have not tried this environment in Windows, however looking at its size (~1.2GB), I do not think it is the whole Neighborhood pack. I do not have a windows machine to test at the moment.

Moreover, it seems CityEnviron binary is still under active development (with it being divided into 001 and 002 etc) https://github.com/Microsoft/AirSim/releases, so I would not expect it to match with the PolyPixel environment. Hope this helps.