microsoft/AirSim #1114: Disable Shadows in CV-Mode

patrickschulte, 2018-05-25:
I try to capture some images for a Deep Learning task. For this i create like in the cv_mode.py example `client = MultirotorClient()
client.confirmConnection()`

I also set the `"UsageScenario": "ComputerVision",` in the settings.json file. But i recognized that i get a shadow of the drone in the scenario, while running it. Is there a possibility to disable this behavior?

Mayankm96, 2018-06-04:
I haven't tried it though but you can disable the cast shadow options in Unreal Engine to do so. 

Source: https://forums.unrealengine.com/development-discussion/rendering/96972-how-to-disable-shadows

patrickschulte, 2018-06-25:
I managed to disable the shadows by opening the drone's .uasset-file (Plugins/Content/Blueprints/BP_FlyingPawn.uasset) in UE4, setting the drone itself to invisible and disabling the casting shadow option. 

Thanks to all!

closed 2018-06-25 by patrickschulte