microsoft/AirSim #1116: Which Unreal Engine is AirSim master branch compatible with?

lburzawa, 2018-05-27:
Linux building instructions talk about only supporting 4.17, however git clone command provided points to branch 4.18. I tried to open AirSim/Unreal/Environments/Blocks/Blocks.uproject with UE4Editor 4.17 and it told me to use newer Unreal Engine. When trying to open it with UE4Editor 4.18 it tells me the project was build with a different Unreal Engine version and says it can rebuild it. Then it says the project could not be compiled and would I like to open it with Null Source Code access. I say yes but nothing happens. Then I try to open AirSim/Unreal/Environments/Blocks 4.18/Blocks.uproject, it says following modules are missing: libUE4Editor-Blocks.so and libUE4Editor-AirSim.so and if I would like to rebuild them. I say yes and then it says "Blocks could not be compiled. Try rebuilding from source manually.". Any advice?

Bryan-bai, 2018-05-28:
Dear,
I have the same problem with you. Have you solved it? Please help me, thanks.

felix-kolbe, 2018-06-06:
Same here. Also tried to open in 4.19 but that doesn't help.

TritonSailor, 2018-06-06:
Try opening Blocks without converting.  It works for me.

sytelus, 2018-06-20:
There was a bug in doc. Please use 4.18 everywhere. If needed you can delete EngineVersion line in uproject and try again.

AloshkaD, 2018-06-23:
My project used to work fine but after updating and recompiling Airsim I got the same problem. @sytelus suggestion didn't work for me. I removed unreal and airsim and compiled them from source in linux but that didn't help.

hashim19, 2018-07-18:
@lburzawa Are you able to solve your problem? I have exactly the same error.

lburzawa, 2018-07-18:
In my case I realized that I was not rebuilding a project with sudo permissions. If you are running on Linux, I believe you can see the exact error in the terminal where you launched "./UE4Editor".