microsoft/AirSim #1121: simPause crash Airsim

gusnitaa, 2018-05-29:
Hello,

I have a problem with the simPause function for the car. During the training of my model, if I use this function (which is important for the quality of the simulations), Airsim crash after some simulations. I tested it in Neighbourhood environment and also in an environment that I created myself, the result is the same, Airsim crash after a few episodes.
My colleague use this function but for the drone and he doesn't have any problem with this.

Can you find out why this function doesn't work properly for the car?

Thank You for your help.

iquater, 2018-05-30:
Hi, i have same problem. I found source of the crash: `void UAirBlueprintLib::setUnrealClockSpeed(const AActor* context, float clock_speed)`. 
Sometimes `GetWorldSettings` function returns nullptr (don't know why).
I just added nullptr check, and now simPause is working without crash. (But i think it's bad fix)

hashikemu, 2018-06-01:
Hi, I have the same problem in simContinueForTime function. Following image is screen when an error occurs.
[image]
I use this function for path following simulation. The timing at which a crash occurs differs every time.
Thanks.

sytelus, 2018-06-23:
I have added above check and also put the code to run on render thread. Still, we don't really know why this may be happening.

jonyMarino, 2022-02-25:
This is not happening in the current version
Closing due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2022-02-25 by jonyMarino