microsoft/AirSim #1125: API Server for each of the Multiple Vehicles spawned

kshitiz38, 2018-05-30:
From my understanding even in the latest release of AirSim, in a multi vehicle scenario there is no way of simultaneously controlling each vehicle through api server running at different ports which is useful when you want the vehicles to interact with each other, each controlled by API in same environment. 

Specifically this abstraction would be more useful when both car and drone are required to be spawned at the same time, like in my case.

This issue is also referenced in #482, but the solution is for a very old release and deprecated.

galgyeony, 2018-06-01:
You may refer to this:
https://github.com/stephen-jang/AirSim
and this:
https://github.com/Microsoft/AirSim/issues/1008

A month ago, a premature (i.e., both vehicles cannot land simultaneously :( ) multi vehicle code was there in the master AirSim branch but now it has been disappeared. Anyhow, that code allocates different port numbers to multiple vehicle. 

I am also waiting for the mature multi-vehicle capability in the near future.

kshitiz38, 2018-06-01:
@galgyeony Yes you can check out my fork. I was able to achieve this multi-vehicle capability.

In current release of AirSim the RpcServerBase.cpp uses a call getVehicleApi() to get the ApiBase of a vehicle but in each GameMode that call only returns the VehicleApiBase for vehicle marked FPV which is only one.

So, what I did was to keep a private variable port in RpcServerBase.cpp and in SimModeBase keep a map of port and VehiclePawnWrapper, which can be easily created in setupVehiclesAndCamera(). Thus whenever getVehicleApi() is called, I simply pass the port along and in GameMode access the map to get the VehiclePawnWrapper corresponding to the port and return its VehicleApiBase.

jixian79, 2018-06-13:
@kshitiz38 your forked project worked well! Thank u very much. Now i can control 3 prones in offcial block environment.

sytelus, 2018-06-23:
It's coming soon: https://github.com/Microsoft/AirSim/issues/1137

closed 2020-02-14 by ironclownfish