microsoft/AirSim #1127: Merging airsim game mode with other game modes

alvikag, 2018-05-30:
Hi,
I am using to airsim for tracking applications which involves using a quadrotor and a ground rover simultaneously. Is there a way such that airsim and carsim can be used at the same time? 
In general, does airsim provide any gamemode merging/overlapping with another so as to have something like keyboard control of an unreal engine ground vehicle along with airsim quadrotor control. 

Thanks
Alvika

DrifterFun, 2018-05-30:
Hi I am now working to merge AirSim and CarSim. However I get some problem now. Maybe we can disscuss on this problem. You can get some information in following link.
[https://github.com/Microsoft/AirSim/issues/1077](url)

kshitiz38, 2018-06-01:
This is also referenced here #506.
You can check out my fork, I was able to achieve this by merging the two game modes.

sytelus, 2018-06-23:
We have done lots of refactoring recently which will make this possible much more cleanly and easily. Now car and multirotor APIs run on same port. Also, there is vehicle_name parameter in all relevant API calls.

aadi-mishra, 2018-07-09:
@sytelus But can you spawn a car and a multirotor with the new repo? I have seen multiple vehicles functionality but all of it seems for similar vehicles. Say multiple quadrotors or cars. Does this make it  possible to import and control both vehicles together? I am currently using Kshitiz's fork.

RVBldr, 2018-10-19:
This seems to be a duplicate of #1382.

FL2316, 2019-05-27:
I follow your advice, but there are some mistakes that can't be solved. I don't know what to do...seg fault,
Can anybody help me?thanks alot

closed 2020-02-14 by ironclownfish