microsoft/AirSim #1128: Fast modification and build on Windows

EthanZhangYi, 2018-05-30:
Hi, I am not familiar with developing on windows by Visual studio. How to do fast modification and compiling for AirSim? (i.e. I want to add some new feature and debug)

Now to use AirSim in an Unreal project, three steps are needed:
(1) compile dependencies such as Eigen, rpclib and copy it into AirLib/
(2) compile AirLib and copy it into Plugins
(3) Compile Unreal project and run it.

How to skip these copying step? Just modify the code of AirSim, build it and run it? Can any official developer give some advice on fast developing and debugging?

saihv, 2018-06-04:
Couple things to note: 

1. IIRC, step 1 happens automatically during build.cmd. And once the deps are built, you wouldn't have to touch them again.
2. Compiling and copying AirLib is only necessary if you're modifying stuff in AirLib: the core part of AirSim such as the firmware and controls. On the other hand, upper level stuff like camera recording, movement, world characteristics etc. are mostly contained in the Plugins folder, and you can change them directly by only compiling the Unreal project. 

Lastly, it might be useful to note that you can always build a VS project (because you'd compile the Unreal project through VS anyway) from commandline. So you can add the respective `msbuild` command to a batch script similar to `build.cmd` so it might save some time for you.

EthanZhangYi, 2018-06-05:
Hi, @saihv  Thanks very much for your reply.
I am wondering if I can directly modify

EthanZhangYi, 2018-06-05:
Hi, @saihv Thanks very much for your reply.
I am wondering if I can directly modify codes in `ROOT_DIR\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib` and compile the Unreal project `Blocks`. (According to this [workflow](https://github.com/Microsoft/AirSim/blob/master/docs/dev_workflow.md), it seems to be allowed.) Or I have to modify codes in `ROOT_DIR\AirLib`, compile by `build.cmd`, copy it to `ROOT_DIR\Unreal\Environments\Blocks\Plugins\AirSim\Source`  and compile the Unreal project.

BTW, besides `AirLib` project, there are several other projects such as `HelloCar`, `DroneServer`, `PythonClient` etc in the AirSim solution. What are the relationship/dependency of these projects? What are they used for?

Regards

saihv, 2018-06-05:
I think the workflow for Blocks is easier because it is structured in a way that it's part of the whole repo, thereby the project. I have only worked with custom environments in the past, so my usual workflow was to copy AirLib to the custom Unreal project after making changes.

HelloCar, DroneServer, PythonClient etc. are all wrapper code that use the internal APIs to perform certain functions. For instance, in DroneServer, you can see various control and access commands being exposed from the Multirotor API (position control, attitude control, image capture and recording): similar stuff can be seen in PythonClient but in a Python interface.