microsoft/AirSim #1132: Auto Driving with Apollo system, the recorded Throttle Steering is always 0

CMXB, 2018-06-05:
When  I am using apollo system to drive the car, the recorded Throttle Steering in airsim_rec.txt is always 0.

CMXB, 2018-06-05:
I fixed it by myself. Change the code in the bottom of  CarPown.cpp   		.append(std::to_string(KPH_int).append("\t"))
		.append(std::to_string(api_->getCarControls().throttle)).append("\t")
		.append(std::to_string(api_->getCarControls().steering)).append("\t")
		.append(std::to_string(api_->getCarControls().brake)).append("\t")
		.append(gear).append("\t");

ll7, 2018-07-18:
Have you been able to connect Apollo from Baidu with AirSim? What was your approach? What kind of interface did you use?

Thank you!

Beijingsheng, 2018-08-20:
How do you add function getCarControls ?  Many thanks