microsoft/AirSim #1134: Is motion platform support available in AirSim?

haider04, 2018-06-05:
Hi,

I wanted to know if some interface to detect motions is present in AirSim? Which could be then fed to a motion platform like D-Box? So that we could have a real feel while driving a car? 

Regards,
Haider

sytelus, 2018-06-23:
One of the contributor had submitted force feedback code for DirectX: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L195

If D-Box etc are available through DirectX Joystick interface then you might be able to use it. Please feel free to contribute :).

RVBldr, 2018-11-21:
Internal VSTF 7429.

hrishikeshtawade04, 2019-04-30:
@sytelus I am using G920 logitech wheel and want to use it with the city map. Will I get the rumble on the wheel by just adding the force-feedback part in the PawnSimApi or do we need to do some additional settings?