microsoft/AirSim #1137: Controlling multiple vehicles from Python

KavehFathian, 2018-06-06:
I've spend a lot of time trying to control multiple vehicles (quadrotors) using the most recent version of AirSim (1.1.10) from Python.  The instruction in "multi_vehicel.md" doesn't work for the "simple_flight" controller; I can only control the first flying pawn from Python.  Is this a bug or is there something wrong with my settings in "settings.json"?  Here are my settings:

    {
    	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    	"SettingsVersion": 1.0,
    	"DefaultVehicleConfig": "SimpleFlight",
    
    	"SimpleFlight": {
    		"FirmwareName": "SimpleFlight",
    		"DefaultVehicleState": "Armed",
    		"ApiServerPort": 41451
    	},
    	"SimpleFlight_1": {
    		"FirmwareName": "SimpleFlight",
    		"DefaultVehicleState": "Armed",
    		"ApiServerPort": 41452
    	}
    }

I would really appreciate any help! Thanks!

P.S. I've read and tried #664 and other multi-agent issues, but non of them worked so far.

galgyeony, 2018-06-07:
You may refer to this:
https://github.com/stephen-jang/AirSim
and this:
#1008

A month ago, a premature (i.e., both vehicles cannot land simultaneously :( ) multi vehicle code was there in the master AirSim branch but now it has been disappeared. Anyhow, that code allocates different port numbers to multiple vehicle. You can control multi vehicles using swarmControl.py from stephen-jang's fork.

Don't be confused; multi_vehicle.md is totally outdated. Setting up settings.json does not work.

I am also waiting for the mature multi-vehicle capability in the near future.

KavehFathian, 2018-06-07:
@galgyeony , I followed your instructions and got it working!  I can't thank you enough!  You have no idea how much time I spent trying to fix the multi-agent functionality in the new version.  I think I'll stick with stephen-jang's fork.  Also, a big thanks to @stephen-jang.

sytelus, 2018-06-23:
We have been doing ton of refactoring to make APIs more clean and add multi-vehicle capabilities. All the core code for this is now pushed and we are testing this capability now.

KavehFathian, 2018-06-23:
Thanks @sytelus !  I'll look forward to using the multi-agent capability in the new version.

galgyeony, 2018-06-26:
@sytelus Thank you so much! By the way, I have a question. Do the new multi-vehicle capabilities work on one machine (like stand-alone mode) or on multiple machines simultaneously (like client-server based network games)?

KavehFathian, 2018-08-02:
@galgyeony Thanks.

I'm running this on a relatively old laptop workstation (Dell Precision M4800).  Here are the specs: 
https://dell.com/en-us/work/shop/dell-laptops-and-notebooks/precision-m4800/spd/precision-m4800-workstation

I've noted the sequence 4->4->1 when the drones were taking off. While sending the motion control commands this was not notable for me and it seemed that they are moving simultaneously. I didn't modify the source code, and this is based on the newest version of AirSim.  If you are having trouble it might be due to using an older version.

yashkaushal, 2018-11-18:
Hi @KavehFathian,

They are both very interesting videos, Can you tell the title of the paper which you've published and is the code opensource?
Also, did you find any way to make them all take off at the same time?

Thanks and Regards
Yash Kaushal

closed 2019-10-03 by msb336