microsoft/AirSim #1138: Setup Environment

tanakataiki, 2018-06-07:
Hello Thanks for a great repo.

I was trying to use visual stdio2017 for AirSim and I typed clean_rebuild.bat in AirSim repo but The following Error Occured. Does anyone met the same error? I am really greatful if you help me.

"C:\git\Airsim\AirSim.sln" (既定のターゲット) (1) ->
"C:\git\Airsim\DroneShell\DroneShell.vcxproj" (既定のターゲット) (2) ->
(Link ターゲット) ->
  AirLib.lib(FileSystem.obj) : error LNK2019: 未解決の外部シンボル __imp_SHGetFolderPathW が関数 "public: static class std::basic_string<char,struct std::char_traits<cha
r>,class std::allocator<char> > __cdecl common_utils::FileSystem::getUserDocumentsFolder(void)" (?getUserDocumentsFolder@FileSystem@common_utils@@SA?AV?$bas
ic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) で参照されました。 [C:\git\Airsim\DroneShell\DroneShell.vcxproj]
  C:\git\Airsim\DroneShell\build\x64\Debug\DroneShell.exe : fatal error LNK1120: 1 件の未解決の外部参照 [C:\git\Airsim\DroneShell\DroneShell.vcxproj]

"C:\git\Airsim\AirSim.sln" (既定のターゲット) (1) ->
"C:\git\Airsim\DroneServer\DroneServer.vcxproj" (既定のターゲット) (6) ->
  AirLib.lib(FileSystem.obj) : error LNK2019: 未解決の外部シンボル __imp_SHGetFolderPathW が関数 "public: static class std::basic_string<char,struct std::char_traits<cha
r>,class std::allocator<char> > __cdecl common_utils::FileSystem::getUserDocumentsFolder(void)" (?getUserDocumentsFolder@FileSystem@common_utils@@SA?AV?$bas
ic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) で参照されました。 [C:\git\Airsim\DroneServer\DroneServer.vcxproj]
  C:\git\Airsim\DroneServer\build\x64\Debug\DroneServer.exe : fatal error LNK1120: 1 件の未解決の外部参照 [C:\git\Airsim\DroneServer\DroneServer.vcxproj]

"C:\git\Airsim\AirSim.sln" (既定のターゲット) (1) ->
"C:\git\Airsim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj" (既定のターゲット) (7) ->
  MavLinkCom.lib(FileSystem.obj) : error LNK2019: 未解決の外部シンボル __imp_SHGetFolderPathW が関数 "public: static class std::basic_string<char,struct std::char_traits
<char>,class std::allocator<char> > __cdecl mavlink_utils::FileSystem::getUserDocumentsFolder(void)" (?getUserDocumentsFolder@FileSystem@mavlink_utils@@SA?A
V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) で参照されました。 [C:\git\Airsim\MavLinkCom\MavLinkTest\MavLinkTest.vcxproj]
  C:\git\Airsim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.exe : fatal error LNK1120: 1 件の未解決の外部参照 [C:\git\Airsim\MavLinkCom\MavLinkTest\MavLinkTest
.vcxproj]

"C:\git\Airsim\AirSim.sln" (既定のターゲット) (1) ->
"C:\git\Airsim\Examples\Examples.vcxproj" (既定のターゲット) (8) ->
  AirLib.lib(FileSystem.obj) : error LNK2019: 未解決の外部シンボル __imp_SHGetFolderPathW が関数 "public: static class std::basic_string<char,struct std::char_traits<cha
r>,class std::allocator<char> > __cdecl common_utils::FileSystem::getUserDocumentsFolder(void)" (?getUserDocumentsFolder@FileSystem@common_utils@@SA?AV?$bas
ic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) で参照されました。 [C:\git\Airsim\Examples\Examples.vcxproj]
  C:\git\Airsim\Examples\build\x64\Debug\Examples.exe : fatal error LNK1120: 1 件の未解決の外部参照 [C:\git\Airsim\Examples\Examples.vcxproj]

sytelus, 2018-06-23:
I'm guessing this is probably due to international version of Windows? You might want to make sure you have installed Visual Studio 2017 with VC++ and Windows SDK 8.1.

tanakataiki, 2018-06-30:
＠sytelus 
I Use
・Airsim1.2.0 binary ver
・X64Nativetools

And Yes I installed 
・VS2017 ver15.7.3
・Windows 10 SDK (10.0.17134.0) 
・Windows 10 SDK forC++(10.0.16・・・
・Windows 8.1 SDK  UCRT SDK
・Cmake3.9.6
Yet not solved build error.

sytelus, 2018-10-17:
We are unable to repro this issue with latest version.

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus