microsoft/AirSim #1140: can I still use vs2015 for UE4.19?

vinbo, 2018-06-07:
My laptop has no extra space for vs 2017. I'm using win7. when I tried to build Airsim as step 3 in this page https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md , the prompt says "Hello there! We just upgraded AirSim to Unreal Engine 4.18 and Visual Studio 2017. Here are few easy steps for upgrade so everything is new and shiny: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md"  But I don't want to update vs2017, how can I do the build step?  Thanks.

sytelus, 2018-06-23:
You can probably use VS2015 if you modify [this line](https://github.com/Microsoft/AirSim/blob/master/build.cmd#L74). Please let us know if that works :).

closed 2018-10-17 by sytelus

xjz89982, 2019-11-20:
@sytelus Hello, i have similar problem with @vinbo . If i can still use vs2015, it helps a lot.
Where should i modify? [this line](https://github.com/Microsoft/AirSim/blob/master/build.cmd#L74) seems to just be a bracket.