microsoft/AirSim #1144: SimGetImage looks very dark

xuhao1, 2018-06-08:
Hi everyone,
I am using SimGetImage to grab image from Unreal Enviroment , but the image looks really dark after I set up all the parameter (like AutoExposureSpeed and so on.)
And also I tried PNG and uncompressed image, nothing changed
[image]

kuanting, 2018-06-12:
I encountered the same issue in several environments, is it something related to Unreal lighting condition?

The screen I saw in Unreal engine
[image]

The screenshot I got using AirSim API
[image]

DavidLSmyth, 2018-06-12:
I found that changing the bp_sky_sphere sun brightness to around 80 made the brightness of the recorded images correspond more closely to the viewport images. Changing the auto-exposure min and max brightness in post-process volume(s) may also be worth looking at from my previous experimentation

sytelus, 2018-06-23:
You might also want to adjust TargetGamma:

```
"CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "TargetGamma": 1.5
      }
    ]
}
```

Also see [note on new settings](https://github.com/Microsoft/AirSim/blob/master/docs/upgrade_settings.md).

mjcorriere, 2018-06-29:
@sytelus Can sensible defaults please be provided such that the output images match the in game images? It is frustrating that additional work needs to be done to capture the exact images that are being shown to the user on screen.

AloshkaD, 2018-12-21:
@kuanting do you mind if I ask you where did you get the drone geometry you have and if you mind sharing the physics script as well. Appreciated!

zimmy87, 2021-03-12:
This looks like it was fixed by #1834, so closing

closed 2021-03-12 by zimmy87