microsoft/AirSim #1147: ROS interface with AirSim

haider04, 2018-06-08:
Hi,

Is there any ROS interface available in AirSim or any API's which could do the work? 
Actually i want to implement a ROS node with AirSim to check the the possibility of running my CV, AI algorithms working in AirSim? 

Any leads to this will be highly appreciated. Thanks!

Regards,
Haider

CaptainJL, 2018-06-09:
Hey,
So i am in the same boat at the moment.
I find that the PX4 (multirotor) system connects via mavros, meaning i can control a quadrotor using ROS.
But images don't naturally come out (which i am trying to solve).
The problem i am not even entirely sure how to use cmake, as all the examples are mostly for VisualStudio.

TritonSailor, 2018-06-09:
You guys can try the sample ROS nodes I contributed a few days ago.  https://github.com/Microsoft/AirSim/pull/1135

The example drone_image_raw.py publishes uncompressed rgba image from airsim as ROS /airsim/image_raw topic.  You may of course make minor edits to the example, to change AirSim camera, image type, etc.

Mayankm96, 2018-06-12:
You may also check the ROS wrapper written by Marcelino Almeida. It can be found out [here](https://github.com/marcelinomalmeidan/publishAirsimImgs). 

I have added the changes he has mentioned in [my fork of AirSim](https://github.com/Mayankm96/AirSim_ROS) which you may directly build and use.

sytelus, 2018-06-23:
I've integrated ROS nodes by @TritonSailor in to master. Note that there have been [API changes in v1.2](https://github.com/Microsoft/AirSim/pull/1135#issuecomment-399330017) so I also upgraded these ROS nodes for new APIs.

@Mayankm96 - it would be great to get this code as PR :).

madratman, 2019-05-13:
Please see [airsim_ros_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_ros_pkgs) and [airsim_tutorial_pkgs](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs) for a ros wrapper over the multirotor API

closed 2019-05-13 by madratman