microsoft/AirSim #1149: Unreal crash Error: Exception occured while updating world: reset() must be called first before update()

srivatsankrishnan, 2018-06-10:
Hi ,
I am trying to run AirSim (master)  in Ubuntu and it seems like Unreal engine(4.18) crashes due this exception:
[2018.06.09-23.48.32:963][980]LogAirSim: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.06.09-23.48.32:963][980]LogAirSim: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.06.09-23.48.32:963][980]LogAirSim: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.06.09-23.48.32:963][980]LogAirSim: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.06.09-23.48.34:295][984]LogAirSim: lookahead = 1.800000, adaptive_lookahead = 1.000000
segment 0 done: x=0.000000, y=0.045924, z=0.045924
overshoot=0.017299

I came across the [#issue 631](https://github.com/Microsoft/AirSim/issues/631) and it looks similar.  Is this fix merged to the current master? 

Thanks
Sri

sytelus, 2018-06-23:
Could you please try latest version. Please note that v1.2 has some API changes so you might have to update your client code. Please see https://github.com/Microsoft/AirSim/blob/master/docs/upgrade_apis.md

elgeom, 2018-09-16:
Also have this problem, the message starts with PositionController error:

[2018.09.16-09.47.36:134][788]LogTemp: Error: Exception occurred while updating world: PositionController does not support yaw axis i.e. 3
[2018.09.16-09.47.36:134][788]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.09.16-09.47.36:134][788]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.09.16-09.47.36:134][788]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2018.09.16-09.47.36:134][788]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()

Wanglonghong222, 2019-09-23:
Also have this problem, the message starts with PositionController error:
Malloc Size=65535 LargeMemoryPoolOffset=196655 
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: PositionController does not support yaw axis i.e. 3
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:141][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:142][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:142][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:142][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:142][439]LogTemp: Error: Exception occurred while updating world: reset() must be called first before update()
[2019.09.23-08.54.01:152][439]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000000

[2019.09.23-08.54.01:152][439]LogLinux: Fatal error!

craig-m-k, 2020-02-11:
I also have this problem. (UE 4.22.3, openSUSE 15.1).  Very similar errors to the ones listed above.

jonyMarino, 2020-09-02:
@craig-m-k did you try with latest version?

jonyMarino, 2020-11-23:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-23 by jonyMarino