microsoft/AirSim #1150: "Error at startup: Error creating directory, hr=5", anyone met this?

vinbo, 2018-06-10:
Hi, there. when I download the binaries and tried to run them, I met the pop up box, "Error at startup: Error creating directory, hr=5", like this
[image]
click ok, and then becomes
[image]
I tried in environment NH, landscapemountain, Zhangjiajie, they all appeared such error. First I thought it was a bug like https://github.com/Microsoft/AirSim/issues/977 , but when I succeed in building the tutorial example https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md, just before the last step---press play button, the error occurred again. 
[image]
I realized that maybe I need to single this out, maybe it's my own problem. (I'm using win7, 64bit, vs2017, ue4.18.3 , i don't know if these matters).

Anyone knows how to solve this? Thanks

vinbo, 2018-06-11:
a new error came out when I run the cityenvironment binaries, after I press the OK of the first error box, then another box popped up, 
[image]
the funny thing is, you can see I don't even I have a D disk label...
so what should I do?

vinbo, 2018-06-14:
@sytelus
any idea what happens here ,please?

vinbo, 2018-06-16:
I think I finally find out the root of the problem, by running the most original version of binaries: AirSimNeighbourhood v1.0.0, although the erro is the same, but more error info was given:
[image]
so the so called "error creating directory" acturally means, the UE or airsim cannot generate the directory to put settings.json, but the things, I'm pretty sure have the admin authority, and even created that find in person, but still changed nothing. so, this time you know how to help me then, right?
I already read the similar issure https://github.com/Microsoft/AirSim/issues/406 , but the difference is, My UE/airsime does not even creat or found that directory, so what can I do, please?
@sytelus @lovettchris

sytelus, 2018-06-23:
In latest version this should not happen. Try 1.2.0 version. The error was due to 30s default time out in Unreal when looping through all objects. In latest versions, we don't have such limitation.

vinbo, 2018-06-23:
Still have the same problem, as the first 2 pics shown in my first post, for the airsimNH1.2. For me the NHv1.0.0 is the best one, i can press F1 for help, and show the 3 subwindow, and change to manual camarer and look around, it is just I cannot move the vehicle, since i don't have a RC at hand, and cannot see the popup dialogue for me to choose car or drone. But for the rest version, like 1.2 or 1.1.10, I can do nothing to change the screen a little except alt+tab to exit.

and for the new binary maze1.2, it's like this
[image]

vinbo, 2018-06-29:
In order to accelerate the solving of the problem, I think I need to add something. Since I can build the demo project blocks "successfully", if I had never press the play button, I could paste the output of the VS here, so you could judge more quicker where the key is,
(environment: win7, 64bit, vs2015, UE4.16, airsimv1.1.10, every thing like the tutorial says) following the steps here https://github.com/Microsoft/AirSim/blob/v1.1.10/docs/unreal_blocks.md , just before press PLAY button, everything seems fine, and things happens like the first post in this issue, as my 3rd pic shows.
If before I press the F5 to run the project, I turn on C++ exceptions from the Exceptions window, as FAQ suggested, this will pop up:
[image]
and then I choose "continue", then this
[image]
choose "continue" again, and becomes
[image]
when I stop the debug, I saved the output of the VS 
[output_window_after_press_all_continueAndok.txt](https://github.com/Microsoft/AirSim/files/2149634/output_window_after_press_all_continueAndok.txt)

If I chose to press "break" not "continue" in the first choice dialog box, then you would see this
[image]
I checked it, and press load button, but it said cannot find the source or sth. then I don't know what to do next.

I hope, the logs here could help you solve it more quicker, thanks

vinbo, 2019-09-08:
same thing happens to v1.2.2, when i tried to run blocks demo. the build part seems normal,  and when i hit the F5 and press the start button of UE, it pops up this: 
[image]
press "ok" of the dialog, it becomes,
[image]
and never move, even armed.
part of the output of vs2017 is:
[2019.09.07-18.00.22:600][532]LogTemp: Repeating last play command: 选中的视口
[2019.09.07-18.00.22:753][532]LogBlueprintUserMessages: Early PlayInEditor Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
[2019.09.07-18.00.22:802][532]LogPlayLevel: PlayLevel: No blueprints needed recompiling
[2019.09.07-18.00.23:495][532]PIE: New page: PIE会话: FlyingExampleMap (2019年9月8日 上午10:00:23)
[2019.09.07-18.00.23:497][532]LogPlayLevel: Creating play world package: /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap
[2019.09.07-18.00.23:909][532]LogPlayLevel: PIE: StaticDuplicateObject took: (0.411299s)
[2019.09.07-18.00.23:910][532]LogAIModule: Creating AISystem for world FlyingExampleMap
[2019.09.07-18.00.23:911][532]LogPlayLevel: PIE: World Init took: (0.001548s)
[2019.09.07-18.00.23:920][532]LogPlayLevel: PIE: Created PIE world by copying editor world from /Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap to /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap (0.422526s)
[2019.09.07-18.00.24:033][532]LogInit: XAudio2 using '喇叭/耳机 (Realtek High Definition Audio)' : 2 channels at 48 kHz using 32 bits per sample (channel mask 0x3)
[2019.09.07-18.00.24:055][532]LogInit: FAudioDevice initialized.
线程 0x1fac 已退出，返回值为 0 (0x0)。
线程 0x17d4 已退出，返回值为 0 (0x0)。
[2019.09.07-18.00.24:200][532]LogLoad: Game class is 'AirSimGameMode'
[2019.09.07-18.00.24:350][532]LogWorld: Bringing World /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap up for play (max tick rate 0) at 2019.09.08-02.00.24
[2019.09.07-18.00.24:351][532]LogWorld: Bringing up level for play took: 0.012949
“UE4Editor.exe”(Win32): 已加载“H:\Epic Games\UE_4.18\Engine\Binaries\Win64\UE4Editor-MovieSceneCapture.dll”。无法查找或打开 PDB 文件。
“UE4Editor.exe”(Win32): 已卸载“H:\Epic Games\UE_4.18\Engine\Binaries\Win64\UE4Editor-MovieSceneCapture.dll”
[2019.09.07-18.00.25:009][532]LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0012 seconds. Added 11 classes and 0 folders.
0x000007FEFD90B87D 处(位于 UE4Editor.exe 中)引发的异常: Microsoft C++ 异常: std::invalid_argument，位于内存位置 0x00000000006BB648 处。
线程 0x2090 已退出，返回值为 0 (0x0)。
[2019.09.07-18.00.46:622][532]PIE: /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap的‘在编辑器中播放’的起始时间23.724
[2019.09.07-18.00.46:634][532]LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel' has LevelScriptBlueprint '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap' with GeneratedClass '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap_C' with ClassGeneratedBy '/Game/FlyingCPP/Maps/FlyingExampleMap.FlyingExampleMap:PersistentLevel.FlyingExampleMap'
线程 0x22e8 已退出，返回值为 0 (0x0)。

(i'm under win7, vs2017(sorry for the chinese output), UE4.18, airsimv1.2.2) any idea, @sytelus ?

WatermelonWarrior, 2020-03-15:
The Similar Problem is solved by adding the file "settings.json" into the folder "C:\Users\USER_NAME\Documents\AirSim"
(my environment: win10, 64bit, vs2017, UE4.18, airsimv1.2.0)
(the location of "settings.json" is AirSim/docker)
Tips: the only dfifference is ERROR#hr=2 for me, but ERROR#hr=5 for you.

closed 2020-09-04 by jonyMarino

lkbhg, 2022-07-29:
You may have changed the default address of the "Documents" in Windows system.

kingchou007, 2023-03-03:
> You may have changed the default address of the "Documents" in Windows system.

Thanks, I have the same issue cause I move airsim to another folder

wsh3011, 2023-09-09:
Did you solve this problem? I met the same problem but do not have any idea about it.

wsh3011, 2023-09-09:
> > You may have changed the default address of the "Documents" in Windows system.
> 
> Thanks, I have the same issue cause I move airsim to another folder

I have exactly the same problem, and I move airsim to another folder in E drive because C drive don't have enough capacity. But I don't know how to fix it. Could you please share about how you solve this problem?