microsoft/AirSim #1154: Sensor Data Coordinate Convention

jameswalmsley, 2018-06-11:
I've read in the documentation that the NED coordinate convention is used throughout AirSim.

Can someone confirm that the sensor data coming into e.g. SimpleFlight is also in NED convention.

1. When looking down on top of the drone, a positive rotation in Yaw (Z down axis) will cause a positive gyro signal?
2. A pitch upwards of the drone (causing the front of the drone to pitch up by a positive angle) will cause a positive gyro signal?
3. A roll to the right will cause a positive gyro signal?

If the drone is upright I can see an acceleration of -9.81 m/s^2
If the drone is rolled 90 degrees on its right, I should see -9.81 on the Y axis?
If the drone is pitched 90 degrees forward (on its nose) I should see -9.81 on the X axis?

Can anyone confirm that my assumptions are correct.

I would really love to be able to pick up the drone and spin it around in the simulator to test this.

Many thanks..

James

sytelus, 2018-06-23:
Yes, I think that's what should happen. Sensor data is also in NED. Try it out and if that's not the case we will look at it.

jameswalmsley, 2018-06-25:
@sytelus Many thanks for getting back to me.

After further investigating it seems to be in NED.
Can you confirm what gravity should look to a NED sensor. If the axis is pointing to the ground, should it see -9.81 ms/s^2 or +9.81?

Currently in AirSim its all -ve, so I assume this is correct. 

I've fed the imu data into the MadgwickAHRS filter, and it also has a NED coordinate system. However I have to negate all the accel data, hence the question.

Thanks

James

kylelindgren, 2019-02-06:
At complete resting state, the IMU should report -9.81 m/s^2 though the AirSim 1.2 API for simGetGroundTruthKinematics() does not account for gravity and it needs to be manually added after converting to body frame.

madratman, 2019-04-30:
Yes, you're right in that simGetGroundTruthKinematics() is in world NED frame and needs to be converted into body frame and gravity needs to be added. See discussion in https://github.com/Microsoft/AirSim/issues/1909. I think you basically did what the first comment about. 
https://github.com/Microsoft/AirSim/pull/1920 adds an IMU API, so I am closing this

closed 2019-04-30 by madratman

yuyangch, 2020-09-28:
Hi, I am confused here,the body frame also uses NED convention right? where z points downward, so if the drone is resting upright, shouldn't the downward pointing  z direction acceleration reads +9.81 instead of -9.81?

namely line 55 of ImuSimple.hpp: 
`output.linear_acceleration = ground_truth.kinematics->accelerations.linear - ground_truth.environment->getState().gravity;`

why subtract ground_truth.environment->getState().gravity    
instead of just adding it?