microsoft/AirSim #1157: Accessing AirSim API through C++ Class in UE4

benjamin-lange, 2018-06-12:
I would like to alter the segmentation setup before capturing data with the "Compute Vision" provided by AirSim.
I am using a custom environment in UE4 18.3 with the current version of AirSim (pulled today).
I am quite new to UE4, so how can I access the AirSim API?
A minimal example would be appreciated, as well as any links or hints.

senakicir, 2018-06-20:
I did something similar to this using interfaces. Here is a tutorial for this: https://wiki.unrealengine.com/Interfaces_in_C%2B%2B

In my case, I needed to access a class in my Unreal project from AirSim. So I made the external class inherit the interface. By doing so, I could use the class' functions within AirSim. 

Good luck!

sytelus, 2018-06-23:
Read the docs: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md

closed 2018-06-23 by sytelus