microsoft/AirSim #1172: How can I set the subwidows for Multiple Vehicles?

jixian79, 2018-06-19:
i have checked the `settings.json` doc for different images for only one vehicle like below 
```
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraID": 3, "Visible": true},
    {"WindowID": 1, "ImageType": 3, "CameraID": 0, "Visible": true},
    {"WindowID": 2, "ImageType": 6, "CameraID": 4, "Visible": true}
  ]
```
but How can I set the subwidows for Multiple Vehicles? I want to set WIN0 to show the first vehicle's view the WIN1 to show the second vehicle's , etc.

jixian79, 2018-06-19:
And i still have one question, the default BP_FlyingPawn i added to the unreal has 4 BP_PIPCamera like the picture below

[image]

should i add the camera to the unreal ?

[image]

galgyeony, 2018-06-19:
As far as I know, all the views (including subwindows) are for the "first" vehicle. So it is impossible to have a view for the other vehicles. (Note that I am using https://github.com/kshitiz38/AirSim and https://github.com/stephen-jang/AirSim for the multi-vehicle capability.)

By the way, which AirSim version (or fork) are using for the multi-vehicle test? If I am correct, the latest version of AirSim does not support multi-vehicle capability.

jixian79, 2018-06-21:
@galgyeony  i am using [kshitiz38](https://github.com/kshitiz38/AirSim)'s version 。And the [video](https://www.youtube.com/watch?v=feL5Exfm3xQ) of the [saihv](https://github.com/saihv/AirSimMulti) version shows the multi views of vehicle. you can click the link to check. I'm not familiar with  c++, so if you can find out how saihv did it i will appreciate it.

jixian79, 2018-06-21:
@saihv

sytelus, 2018-06-22:
I recently did lots of refactoring to add multi-vehicle capabilities in AirSim. Now most of the infrastructure is there and its only a matter of testing thins out. 

Subwindows currently doesn't allow to specify vehicle_name but it can be easily added. Notice that in new APIs (since v1.2, release last week), there is now an extra parameter to take vehicle_name. New vehicles can be spawned by specifying in settings with names assigned to them.

saihv, 2018-06-22:
@galgyeony @jixian79

https://github.com/saihv/AirSim/blob/develop/Unreal/Plugins/AirSim/Source/Recording/RecordingThread.cpp

This was my version of doing multi vehicle recording, but this is from more than 6 months ago. Please note that the API might have changed significantly, but hopefully this gives you an idea of how the idea works. Essentially instead of passing one camera object, I pass a vector of them and iterate over all cameras when recording. This is also not very efficient because the rendering would happen sequentially. 

If someone can find a better way to do multi camera recording that would be great! (IIRC, the Unreal car sim Deepdrive claims to do very fast multi camera recording using some special low level libraries)

PR #2841 (2020-09-03): Add Vehicle option for Subwindow settings
Example Settings - 
```
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,
    "SimMode": "Car",

    "Vehicles": {
        "Car1": {
          "VehicleType": "PhysXCar",
          "AutoCreate": true
        },
        "Car2": {
          "VehicleType": "PhysXCar",
          "AutoCreate": true,
          "X": 0, "Y": 10, "Z": 0
        }
    },
    "SubWindows": [
        {"WindowID": 0, "CameraName": "0", "ImageType": 3, "VehicleName": "Car1", "Visible": false},
        {"WindowID":  [... cut]
closed 2020-09-03 by madratman