microsoft/AirSim #1174: how to convert the position and rotation in unreal coordinate to airsim coordinate

husha1993, 2018-06-19:
Hi,

The instructions here https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#coordinate-system
show how to convert the positions in ue4 to airsim.
May I know how to convert rotation from ue4 to airsim?
is the x,y,z axis in ue4 coordinate system parallel to those of airsim ?

husha1993, 2018-06-19:
And what is the pawn position/ start position of the drone in ue coordinate system?
It seems that the PlayerStart position is not the drone start position.

sytelus, 2018-06-22:
See here for how to do UE to NED transform: https://github.com/Microsoft/AirSim/blob/v1.2.0/Unreal/Plugins/AirSim/Source/NedTransform.cpp#L49

Yes, X, Y, Z is aligned by Z is inverted.

Pawns are spawned at Player Start.

husha1993, 2018-06-24:
Thanks!
From https://github.com/Microsoft/AirSim/blob/aca1ff810ca7d8019ec43f5f0db39d6a633c8ba4/Unreal/Plugins/AirSim/Source/NedTransform.cpp#L31 ,
I understand that in order to convert the position from ue4 to airsim NED, the "local_ned_offset_" is needed.

https://github.com/Microsoft/AirSim/blob/aca1ff810ca7d8019ec43f5f0db39d6a633c8ba4/Unreal/Plugins/AirSim/Source/NedTransform.cpp#L23
the "local_ned_offset" is related to the pawn start position in ue4 coordinate system,  how can I know the pawn start position from airsim API?

closed 2018-10-17 by sytelus