microsoft/AirSim #1177: Build Complete with VS2017 but Unreal/plugin/AirSim folder not populated

ganscool86, 2018-06-20:
Hi,

I just pulled the code from GiTHUB and ran build.cmd in x64 Native Tools command prompt for VS 2017. The build ran successfully without errors but the folder Unreal/Plugins/AirSim folder contains only: Content, Source and AirSim.uplugin. It seems binaries folder is not created. Can anyone suggest what is wrong? My build ended as captured in the screenshot. Thanks.  
[image]

---- (Remove above before filing the issue) -----

NextSim, 2018-06-20:
I believe the content and source folders and the .uplugin, should be all you need.  Drop the Plugins/AirSim folder into an unreal project and use AirSim with that environment.  Follow the instructions for adding the AirSim plug in and you'll be all set, or use the blocks environment.

https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md

KenChanKH, 2018-06-21:
Congrats in building the AirSim source completely. I know it's out of this thread, but is it possible to share the tricks you use to complete the installation? Thanks a lot

NextSim, 2018-06-22:
If you're having trouble building I'd recommend creating a new thread, detailing what OS you are using, what version of Visual Studio, etc, and what issue you are having.

KenChanKH, 2018-06-23:
Hi NextSim,
Thanks for replying. I did open a new thread on my issue: https://github.com/Microsoft/AirSim/issues/1184
Appreciate it if you can help me on this :)
Thanks :)

bertram1isu, 2018-06-23:
KenChanKH, I just went through getting the build working on Windows and also ran into some issues (different ones than you it looks like).  I captured an issue for it #1190.

This pull request has my proposed changes to the Windows build instructions.  There may be a useful nugget in there for you as well.
https://github.com/Microsoft/AirSim/pull/1191

sytelus, 2018-06-23:
Just tried out fresh clone and I'm able to run everything end-to-end without any issues.

jonyMarino, 2020-09-02:
Closed due to age.
Please feel free to open a new issue (refering this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-09-02 by jonyMarino