microsoft/AirSim #1178: rpclib: Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

hassanyf, 2018-06-20:
I am using VS 2017 and Unreal Engine 4.18.3. When I type "build.cmd", I get the following error:

```
D:\AirSim>build.cmd
Found cmake version: 3.11.4

********************************************************************************
*********
Downloading rpclib
********************************************************************************
*********
Exception setting "SecurityProtocol": "Cannot convert null to type "System.Net.
SecurityProtocolType" due to invalid enumeration values. Specify one of the fol
lowing enumeration values and try again. The possible enumeration values are "S
sl3, Tls"."
At line:1 char:32
+ & { [Net.ServicePointManager]:: <<<< SecurityProtocol = [Net.SecurityProtocol
Type]::Tls12; iwr https://github.com/rpclib/rpclib/archive/v2.2.1.zip -OutFile
external\rpclib.zip }
    + CategoryInfo          : InvalidOperation: (:) [], RuntimeException
    + FullyQualifiedErrorId : PropertyAssignmentException

The term 'iwr' is not recognized as the name of a cmdlet, function, script file
, or operable program. Check the spelling of the name, or if a path was include
d, verify that the path is correct and try again.
At line:1 char:89
+ & { [Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]:
:Tls12; iwr <<<<  https://github.com/rpclib/rpclib/archive/v2.2.1.zip -OutFile
external\rpclib.zip }
    + CategoryInfo          : ObjectNotFound: (iwr:String) [], CommandNotFound
   Exception
    + FullyQualifiedErrorId : CommandNotFoundException

The term 'Expand-Archive' is not recognized as the name of a cmdlet, function,
script file, or operable program. Check the spelling of the name, or if a path
was included, verify that the path is correct and try again.
At line:1 char:19
+ & { Expand-Archive <<<<  -Path external\rpclib.zip -DestinationPath external\
rpclib }
    + CategoryInfo          : ObjectNotFound: (Expand-Archive:String) [], Comm
   andNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Could Not Find D:\AirSim\external\rpclib.zip
Unable to download high-polycount SUV. Your AirSim build will use the default ve
hicle.

#### Build failed - see messages above.

D:\AirSim>
```

Can anyone provide me with a solution?

sytelus, 2018-06-23:
Just tried doing clean clone and can't reproduce this error. I would suggest to check your Internet connection.

sytelus, 2018-10-17:
We are unable to repro this issue with latest version.

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus