microsoft/AirSim #1179: Can't ping the server + server version doesn't exists

Ricocotam, 2018-06-20:
I'm proposing a potential solution to #609 and #1141 (at least). When using the new API (master) on the releases we can't connect as CarClient() :

```
from AirSim.PythonClient.airsim.client import CarClient

client = CarClient()
client.confirm_connection() # crash WSAECONNREFUSED
```

The issue is caused by the pinging here (in AirSim/PythonClient/client.py, line 53) :

```
def confirmConnection(self):
    if self.ping():
        print("Connected!")
    else:
        ....
```

The solution lies in the initialisation of the CarClient. According older versions, it should be on port 42451 and Motorclient on 41451. But in the AirSim/PythonClient/client.py file the port for the ClientCar is not given and thus use 41451 adress so it fails. Solution to problem 1, set the port on line 260 :

```
class CarClient(VehicleClient, object):
    def __init__(self):
super(CarClient, self).__init__(port=42451)
```

Second problem. Even when doing this, we have another issue. New API asks to the server what is its version. The thing is the released servers don't have this command so it fails when asking to the server. I don't know simple solution but a new release.

sytelus, 2018-06-22:
Hi @Ricocotam - we have released AirSim 1.2 with new APIs. These APIs will not work with binaries created before 1.2 and you will indeed see above issues. Now we have also released [binaries for AirSim 1.2](https://github.com/Microsoft/AirSim/releases/tag/v1.2.0). So please get new binaries (or run from the source code) and use latest APIs and everything should work!

closed 2018-06-27 by Ricocotam

vkolhekar, 2020-04-02:
@sytelus can you please provide a link to AirSimTensorflow python APIs that are compatible with 1.2?
I am getting the error.

HareshKarnan, 2020-12-15:
same error here. has anyone found a solution to this issue ?