microsoft/AirSim #1183: moveToPositionAsync() cannot work

husha1993, 2018-06-21:
Hi,

simSetVehiclePose() can work while moveToPositionAsync() cannot work.

'''
currentpose = client.simGetVehiclePose()
currentposition = currentpose.position
currentorientation = currentpose.orientation
print(currentpose.position)
step = [0, 0, 1]
client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(currentposition.x_val + step[0],currentposition.y_val + step[1],currentposition.z_val + step[2]), airsim.to_quaternion(0,0,0)), True)
'''
#simSetVehiclePose() can work

'''
currentpose = client.simGetVehiclePose()
currentposition = currentpose.position
currentorientation = currentpose.orientation
print(currentpose.position)
step = [0, 0, 1]
client.moveToPositionAsync(currentposition.x_val + step[0],currentposition.y_val + step[1],currentposition.z_val + step[2],5)
'''
# moveToPositionAsync() make the python console hang

husha1993, 2018-06-21:
and sometime simSetVehiclePose() cannot work
'''
>>> step = [0, 0, -20]
>>> client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(currentposition.x_val + step[0],currentposition.y_val + step[1],currentposition.z_val + step[2]), airsim.to_quaternion(0,0,0)), True)
'''
If I type those commands in console continuously, sometimes the drone can move to a higher position, while sometimes it keep still.

vxgu86, 2019-07-03:
I have the same problem in HIL simulation.
still not solved , did you get the solution? 
https://github.com/microsoft/AirSim/issues/2055