microsoft/AirSim #1184: Problem building AirSim using build.cmd

KenChanKH, 2018-06-21:
Hi,

I'm new to AirSim and I'm facing issues building the AirSim codes using build.cmd. My system is as follows:
1. Windows 10 Pro 64-bit with the latest updates
2. MS Visual Studio 2017 Community
3. CMake 3.10, installed when build.cmd prompted to install CMake

When I ran build.cmd using the x64 Native Tools Command Prompt, it prompted me that the CMake version is not correct. I followed the instruction to download and install CMake. I have also checked that CMake is set in the Environment Path.

Through this forum, I found a solution (?) where I can bypass the constant checking and downloading of CMake and proceed to the download and install of rpclib. However, when comes to the Build rpclib step, the following error appears:

[image]

What should I do?

Thanks in advance!

sytelus, 2018-06-22:
May be you didn't installed VC++ with your VS 2017?

https://github.com/Valloric/YouCompleteMe/issues/2945#issuecomment-372122560

KenChanKH, 2018-06-23:
Hi sytelus,

I have installed VC++ with my VS2017 installation. I tried the modifications as suggested in your reference, but no change. What else could go wrong? Thanks :)

bertram1isu, 2018-06-23:
In VS2017, if you go to 'Tools->Get Tools and Features...' and you explode the 'Desktop Development with C++' in the summary pane on the right, is 'Visual C++ tools for CMake' checked?

sytelus, 2018-06-23:
May be try installing latest cmake for Windows?

KenChanKH, 2018-06-24:
Perhaps, I give a bit more detail on what I had tried.

1. I had tried reinstalled CMake (v 3.10) and is in my C:\Program Files directory. The Environment PATH is also added in Windows.
2. I had also check "Visual C++ tools for CMake" is enabled in my VS2017. My VS2017 is installed in C:\Program File (x86)
3. When I restarted my laptop and re-run build.cmd without commenting out the check on the CMake version, it still prompt me to install CMake 3.9, but my system has already installed CMake 3.10.
4. When I commented out the check CMake steps in build.cmd (using Wordpad), it continue with the installation until it comes to building rpclib, and it still shows the errors I faced in my first post.

Appreciate any more suggestions to help me move forward.

Thanks in advance!

sytelus, 2018-10-17:
We are unable to repro this issue with latest version.

If this is still a problem, please file a new issue with code, settings and clear steps to reproduce the issue.

closed 2018-10-17 by sytelus