microsoft/AirSim #1185: Framerate issues on Ubuntu 18.04

bertram1isu, 2018-06-21:
Hi all,

I'm wondering if anyone else has tried getting AirSim working on an ubuntu 18.04 machine.  I've followed all the directions I could find in the startup guides for both AirSim and UE, and it does work, but the frame rates are very slow (<10 fps, I'd guess).

I'm on a PC with a Nvidia Titan Xp running version 390.48 drivers installed via the 'update-drivers' ubuntu command.  With the Linaro glmark2 benchmark, I'm able to verify it's using the Nvidia driver and not a software rendering like mesa, seeing FPS above 10,000-20,000 FPS (!) on most of the benchmarks, so I don't think I have a drivers performance issue.

What I see when I start the Blocks environment and select quadcopter is the quadcopter show up and from the rotating of the propellers, I would guess a reasonably high frame rate (60 fps maybe?). 

When I connect the hello_drone python client, I see an immediate slowdown of the propellors and drop in frame rate.

I see similar results when I run the sim within UE or if I run it as a separate process via the play button dropdown selection.  This may just be an issue of not understanding how to build a non-debug / well performing version of an UE simulation, and if so, apologies!

Would just like to get a sense of what is the expected framerates that are normally found with AirSim and whether others are seeing something similar.  I'm walking through the settings looking for things it might be... also searching through forums for both UE and AirSim, but so far no dice.

Thank you!
- Josh.

sytelus, 2018-06-22:
Turn off ["Use CPU..." setting](https://answers.unrealengine.com/questions/474269/keeping-editor-running-as-normal-when-not-in-focus.html) in UE editor!

bertram1isu, 2018-06-23:
That worked, thanks!

Can we add that to the Linux build instructions and/or usage instructions?  (Or is that something I can fix and offer as a pull request?)

- J.

nightduck, 2020-04-21:
I've been seeing this issue on the latest version of airsim and unreal editor. The "Use less CPU..." fix doesn't nothing.