microsoft/AirSim #1188: Windows build instructions may benefit from more detail

todfrye, 2018-06-22:
Noticed two steps i had to work through before i could get this to build.
Builds just fine after i address these.

1) i was getting a compile error - turns out i needed Windows SDK 8.1 installed:
 c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.10.25017\include\crtdefs.h(10): 
fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory [C:\AAC\AirSim\MavLinkCom\MavLinkCom.vcxproj]

2) i needed to install the Unreal Engine Installer:
open your visual studio installer, choose Modify, choose Individual Components, choose Unreal Engine Installer under "Games and Graphics" section. Install this component should solve your problems.

sytelus, 2018-06-22:
Thanks for the feedback. I've updated the build instructions. However, we already had Unreal Engine installation instructions there so you shouldn't have needed 2nd step.

closed 2018-06-22 by sytelus

james-mchugh, 2018-11-21:
I also had too install Unreal Engine through VS despite already having it installed through the Epic Games Launcher.