microsoft/AirSim #1194: Some functionality is needed to accelerate the reinforcment learning time

vinbo, 2018-06-24:
As a platform of developing AI algorithm, airsim seem a great choice. But if I want to develop my reinforcment learning algorithm, the tutorial showing here, https://github.com/Microsoft/AirSim/blob/master/docs/reinforcement_learning.md, and https://youtu.be/fv-oFPAqSZ4 , looks great, but the visuralization part is not necessary to show them all the time along the whole learning process. We all know that a million episodes learning process is really common in RL, but the visuralizaion part costs too much computation resources and time. So do we have some way to choose not to show the 3D animation, during the learning process, which could really accelerate the learning speed and save us so much time? Thanks.

sytelus, 2018-06-24:
Look for NoDisplay ViewMode in [settings](https://github.com/Microsoft/AirSim/tree/master/docs/settingd.md). This allows you to turn off main rendering. Also look for pause and continue APIs which allows you to run simulation at max possible speed.

vinbo, 2018-06-24:
Thanks for such quick reply on sunday. Squeeze the main window is far from what I want, I want to squeeze them all for a certain long time, which could really speed up my running time.

vinbo, 2018-06-24:
oh, I'm so stupid, I can choose not to show the subwindows. So the problem solved,thanks

closed 2018-06-24 by vinbo

LevineYang, 2018-09-30:
can you share the link for settings again, @sytelus . 404 pops up when I try to open it. Thank you!