microsoft/AirSim #1201: Ubuntu Install broken

nicovanduijn, 2018-06-26:
While installing AirSim on Ubuntu 16.04, I encountered an error. 

When trying to open the "Blocks" project, I get prompted that the modules "libUE4Editor-Blocks.so" and "libUE4Editor-Airsim.so" are missing. If I choose to build them, the file

/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/include/common/AirSimSettings.hpp:602:

thows an

error: no viable conversion from
      returned value of type 'unique_ptr<msr::airlib::AirSimSettings::PX4VehicleSetting,
      default_delete<msr::airlib::AirSimSettings::PX4VehicleSetting>>' to function return type 'unique_ptr<msr::airlib::AirSimSettings::VehicleSetting,
      default_delete<msr::airlib::AirSimSettings::VehicleSetting>>'
        return vehicle_setting;
               ^~~~~~~~~~~~~~~

Anyone else encountered this before?

TritonSailor, 2018-06-26:
I'm seeing the same error on my Ubuntu 16.04 laptop, however Blocks built successfully on my Ubuntu 16.04 workstation.  (We've got some lovely new blocks in the level)

I'm trying to determine what is different on my two machines.  I tried copying the working Blocks project to my laptop, but the same error occurred when UE4 loaded and rebuilt the project.

nicovanduijn, 2018-06-26:
I reverted back to v1.1.9 and the error is gone. @TritonSailor Are you using the same version on both machines?

TritonSailor, 2018-06-26:
Yes, using latest master on both machines.  That's why I mentioned the lovely new blocks.  There are a few colored blocks in the new version.

sytelus, 2018-06-27:
I'm not seeing this issue but I thought fix was obvious so just made the change anyway. This is likely [Clang bug](https://stackoverflow.com/questions/36752678/clang-returning-stdunique-ptr-with-type-conversion).

nicovanduijn, 2018-06-27:
Thanks for the fix. I'm getting a different error now, though:

```
The project could not be compiled. Would you like to open it in Null Source Code Access?

Running /home/nico/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/home/nico/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Scale at X=65.000000, Y=24.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Scale at X=1280.000000, Y=720.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Scale at X=1280.000000, Y=720.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Setting swap interval to 'Immediate'

Running Mono...

~/UnrealEngine/Engine ~/AirSim
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
```
And in the compilation, SimJoyStick.cpp failed with:
```
/home/nico/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:252:9: error: use of undeclared identifier 'unused'
        unused(maps);
        ^
/home/nico/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:326:9: error: use of undeclared identifier 'unused'
        unused(index);
        ^
/home/nico/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:327:9: error: use of undeclared identifier 'unused'
        unused(strength);
        ^
/home/nico/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:332:9: error: use of undeclared identifier 'unused'
        unused(index);
        ^
/home/nico/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:333:9: error: use of undeclared identifier 'unused'
        unused(strength);
        ^
5 errors generated.
[7338/8742] Compile
```

nicovanduijn, 2018-06-27:
#1205 addresses the above mentioned problem. Can close this when it gets merged

commit b9f56e00 (2018-06-28): Merge pull request #1205 from nicovanduijn/master

Add missing  include for unused() function, Fixes #1201
closed 2018-06-28 by sytelus