microsoft/AirSim #1207: Building AirSim on Ubuntu 14.04

ThorThorson, 2018-06-27:
Hey there,

for compatibility reasons I'm using Ubuntu 14.04. I've successfully built Unreal Engine and it works. Now I'm trying to build AirSim, but when I run the setup.sh, clang-5.0 is required. Installing clang following these instructions doesn't work because of various missing dependencies that are not installable. 
I'm not very experienced and looking for someone to tell me which lead I can follow to successfully install AirSim. Do I need clang-5.0 or does it work with 3.9 or 3.8 as well and how?

Thanks!

vinbo, 2018-06-28:
actually i met the same problem before, and succeed in install clang 5.0 manually, but failed on installing opengl 4.3 under this environment, so I gave up ubuntu 14 finally, and went up to 16,  of course, things also depends on your hardware, so you need to report your graphic card, and check out the possibility

sytelus, 2018-06-28:
AirSim does install clang 5 automatically when you run setup.sh. Is that not working?

See line: https://github.com/Microsoft/AirSim/blob/master/setup.sh#L49

vinbo, 2018-06-28:
I forgot to mention I am using airsim v1.1.10, not master version, so see this line
https://github.com/Microsoft/AirSim/blob/v1.1.10/setup.sh#L41
can you imagine my feeling  when I saw the setup out uninstalling my clang 5.0 and installing clang3.9?

ThorThorson, 2018-06-29:
@sytelus yes, it tries to install clang 5, but it doesn't exist in standard repos for 14.04. After adding an additional repo manually, I get the following output (this is only th end) when trying to install AirSim. Do you have any idea how to solve that or easily install all the dependencies? 

@vinbo I feel you :D 
Did v.1.1.10 work with clang 3.9 after all?  

+ sudo apt-get install -y clang-5.0 clang++-5.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'clang-5.0-examples' for regex 'clang++-5.0'
Note, selecting 'python-clang-5.0' for regex 'clang++-5.0'
Note, selecting 'clang-5.0' for regex 'clang++-5.0'
Note, selecting 'libclang-5.0-dev' for regex 'clang++-5.0'
Note, selecting 'clang-5.0-doc' for regex 'clang++-5.0'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 clang-5.0 : Depends: libjsoncpp1 but it is not installable
             Depends: libllvm5.0 (>= 1:5.0~svn298832-1~) but it is not going to be installed
             Depends: libstdc++6 (>= 5.2) but 4.8.4-2ubuntu1~14.04.4 is to be installed
             Depends: libstdc++-5-dev but it is not installable
             Depends: libgcc-5-dev but it is not installable
             Depends: libobjc-5-dev but it is not installable
             Depends: libclang-common-5.0-dev (= 1:5.0.2~svn328729-1~exp1~20180509124008.99) but it is not going to be installed
             Depends: libclang1-5.0 (= 1:5.0.2~svn328729-1~exp1~20180509124008.99) but it is not going to be installed
             Recommends: llvm-5.0-dev but it is not going to be installed
             Recommends: libomp-dev but it is not installable
 libclang-5.0-dev : Depends: libstdc++-5-dev but it is not installable
                    Depends: libgcc-5-dev but it is not installable
                    Depends: libobjc-5-dev but it is not installable
                    Depends: libclang1-5.0 (= 1:5.0.2~svn328729-1~exp1~20180509124008.99) but it is not going to be installed
                    Depends: libclang-common-5.0-dev (= 1:5.0.2~svn328729-1~exp1~20180509124008.99) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

vinbo, 2018-06-29:
yes, my v1.1.10 works fine by now.
sorry i forgot how i install the clang5 myself, i googled several ways, and one of them helped.

ThorThorson, 2018-06-29:
But that is on Ubuntu 16.04 now, right? 
Unfortunately I definitely need to stay on 14.04...

vinbo, 2018-06-29:
I was talking about ubuntu 14.04, clang5.0, I did it someway, by the help of google...

yet another extrem way i can think of now, is to modify the setup.sh and build.sh, so you can using the highest version of clang you can install, but I don't know the consequence, what do you think it will happen?  @sytelus

jonyMarino, 2020-10-19:
The current recommended and tested environment is Ubuntu 18.04 LTS. This PR #3067 has instruction for 20.04

closed 2020-10-19 by jonyMarino