microsoft/AirSim #1209: Help:the Blocks project build error

nudtgjx, 2018-06-28:
I pull the latest master branch and navigate to folder AirSim\Unreal\Environments\Blocks and run update_from_git.bat.  Then open Blocks.sln in Visual Studio 2017 and build configuration is set to DebugGame_Editor and Win64. Build this project and the errors as follow:

[image]
Window 10
Visual studio 2017 Community Version
Unreal 4.18.3

NextSim, 2018-06-28:
Did you build airsim first?

You will also need to select "Blocks" in the solution explorer of the blocks environment and set it as the startup project.

nudtgjx, 2018-06-29:
According to your suggestiong, I build the project of Airsim. Then there is the new error as shown as the figure.
[image]
And I igore it and continue to run the clean.bat in the Blocks environment and rebuild the blocks project as the same as before. But the previous errors still exists.

NextSim, 2018-06-29:
Did you build through visual studio, or the VS command prompt?

Follow the instructions here:
https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md

nudtgjx, 2018-06-29:
Yes, I just did it according to https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md. I built projects successfully except the UnrealPluginFiles project. And I build the blocks environment and the errors still occured. Thanks a lot for your help.

sytelus, 2018-07-02:
Please copy and paste text from Output window. It's hard to see things in screenshots.

nudtgjx, 2018-07-08:
The build enviroment as follow:
Window 10
Visual studio 2015 Community Version
Unreal 4.19
Airsim 1.1.1

sytelus, 2018-10-17:
One possibility is unicode characters in path is throwing off some code.

closed 2020-04-06 by madratman